mpaland/printf #1: why *.cpp and not *.c

mgiaco, 2017-11-09:
Is there any special reason why you use cpp?

many thanks

mpaland, 2017-11-09:
@mgiaco Thanks for your interest!
It needs to be a C++ module in the moment cause it contains a templated `_ntoa()` function for number to ASCII conversion.
This is used/called with six different types (`int`, `long`, `long long` and in their unsigned form).
Basically a non templated standard `_ntoa()` implementation would be fine and perhaps better regarding compiled code size.

I leave this issue open and think about it.

mgiaco, 2017-11-09:
Do you have any plans to Support C only?

mpaland, 2017-11-13:
@mgiaco I created a no_ntoa_template branch which contains a pure non templated print.c version.
All tests are passing.
Please test this. If everything is fine, this will be merged into master.

closed 2017-11-21 by mpaland

mgiaco, 2017-11-22:
hey sorry i was on holiday, many thanks i will test it

mpaland, 2017-11-23:
No problem. The non-templated .c version is already merged and successfully tested.
If you encounter any problems, pls open a new issue.

commit 7936fc34 (2017-11-21): Changed printf.cpp to printf.c to be usable with 'C'-compilers

closes #1