mpaland/printf #2: First Test works great

mgiaco, 2017-12-07:
Hi, 

Today I made at first test on my stm32 board with your lib. First impression works great an compiles without any warning on gnu_arm_gcc 5.x

One usage question... normaly I use :
```
printf("\nHallo Welt");
```
if I would like to print something at the beginning of a new line. But with your implementation I do also need some additional line feed - otherwise the line does not start at position 0 on a terminal:
```
printf("\n\rHallo Welt");
```
So I am wondering if the _printchar should handles this is the printf implementation responsible for that?

many thanks

mpaland, 2017-12-07:
Hello @mgiaco ,
printf is just the string "transporter" for sprintf, which writes the formatted text to a given buffer.
So printf creates its own temp buffer of PRINTF_BUFFER_SIZE size, formats the given text and calls `_putchar()` char by char for output.
Any text without the format specifier (%) is copied directly into the output (string) like\n \r etc.

_putchar and printf have no idea how your terminal looks like. Some terminals just need a CR, others (like yours) need a CRLF to go the next line.
As you suspected, this is (normally) all handled in your custom `_putchar()` routine.
Something like:
```c
void _putchar(char character)
{
  if (character == '\n') {
    send('\n');
    send('\r');
  }
  else {
    send(character);
}
```
`send()` is your terminal/serial/whatever output here.
Other terminals may need a translation to ESC sequences or something else...

Same if you implement \t \v tab support. If `_putchar()` processes this sequence you have to move the cursor to the next horizontal or vertical tab position with the emulation codes you use.

mgiaco, 2017-12-11:
okay many thanks.

closed 2017-12-11 by mgiaco