mpaland/printf #3: Can you made it more configurable?

mgiaco, 2018-04-10:
Hi,

On small targets I do not use Float, Long and I also do not need a big buffer size. Can you made the code more configurable. I want to use the source as submodule as it is without any changes if possible. So it would be gread if you can add this or something else which solves the problem

```C

// buffer size used for printf (created on stack)
#ifndef PRINTF_BUFFER_SIZE
#define PRINTF_BUFFER_SIZE    128U
#endif

// ntoa conversion buffer size, this must be big enough to hold one converted numeric number (created on stack)
#ifndef NTOA_BUFFER_SIZE
#define NTOA_BUFFER_SIZE      32U
#endif

// ftoa conversion buffer size, this must be big enough to hold one converted float number (created on stack)
#ifndef FTOA_BUFFER_SIZE
#define FTOA_BUFFER_SIZE      32U
#endif

// define this to support floating point (%f)
#ifndef PRINTF_NO_FLOAT_SUPPORT
#define PRINTF_FLOAT_SUPPORT
#endif

// define this to support long long types (%llu or %p)
#ifndef PRINTF_NO_LONG_LONG_SUPPORT
#define PRINTF_LONG_LONG_SUPPORT
#endif

```

so no I can use

```C
#include "printf_config.h" 
#include "printf.h"
```
In the printf config i use

``` C
#define PRINTF_BUFFER_SIZE 32
#define PRINTF_NO_FLOAT_SUPPORT
#define PRINTF_NO_LONG_LONG_SUPPORT
```

many thanks
cheers
mathias

mpaland, 2018-04-10:
My original intention was to make the main module configurable just inside the module.
But of course you are right, using the printf module as shared, versioned file/lib prohibits that.
A "print_cfg.h" somewhere in the include path seems to be the most reasonable solution.
I'll have a look...

mgiaco, 2018-04-11:
Yes that would be a nice solution.
many thanks...

mgiaco, 2018-04-11:
I have added a pull

mpaland, 2018-04-12:
@mgiaco Thanx a lot for the PR! I had already created a modification.
Please see the `print_cfg` branch. If this is okay for you, I merge it into master and create a new version.
Pls test and give me feedback.

mgiaco, 2018-04-12:
That dose not solve the problem. Now you have added a printf_cfg.h ... so when i use the lib as submodule or as subtree i need to modify the header so there is no real difference to the master.  With my pull if I do not want to change anything everything works the same as in the master but if i would like to change some defines i can add them and leave all that in my main project. So your printf lib can be fetched and leaves untouched.

mgiaco, 2018-04-13:
It's just here in the same folder for delivery, yes and that is the point. Can you remove that. Because now I can not use print as an git submodule from your repo as an example.

mgiaco, 2018-04-13:
I mean if you do not add this config file to the master branch ever thing is great. In the master the only place where the config should be is in the test folder.

mpaland, 2018-04-13:
All right, @mgiaco, now I see what you are after!

Yes, a config file in the project folder *can* cause problems here, but not if your include path has a cfg folder first. But honestly, I won't trust that.

Anyway, embedding foreign projects via links/references/submodules caused me **a lot** of trouble in the past. As you can see, printf is using the catch framework, so I copied the according file(s) and maintain them by myself. (update to actual catch 2.2.2 is coming)
So everything is nice, clean and cherrypicked.
Have a look in my old [sqlitepp](https://github.com/mpaland/sqlitepp) project. sqlite3 is directly bound to the project.
[Here](https://blog.github.com/2016-02-01-working-with-submodules) is an interesting article about (not) using submodules.

In your PR you rely on a compiler switch (PRINTF_CONFIG) which should enable config file inclusion. This switch needs to be defined as compiler parameter (-dPRINTF_CONFIG). That may cause problems in bigger projects if you have a lot of libs which use such technique. I'm trying to avoid that due to fail safe reasons.

Let me think about the options here...

mgiaco, 2018-04-13:
Okay thanks I can live with my fork no problem ... I thought maybe you will add it back in any way. 
A switch is better than no option to change anything :-) 

Submodules are great but dangerous if you do not know what you're doing but I like it more than subtrees. I do have a lot of projects dealing with the same libs so therefore cherry picking is no option because then I have to do a lot of maintenance work on each project.

closed 2018-04-13 by mgiaco