mpaland/printf #11: Return value off by one

Leandros, 2018-04-25:
The return value should return the number of bytes written, without the terminating zero (as mentioned by the printf spec and your source code comments).
Although, the following returns 10, instead of 9:

```
char buf[10];
int size = snprintf(buf, 10, "hello, world");
/* size is 10, but should be 9. */
```

mpaland, 2018-04-25:
Hi Arvid, thanks for your feedback.
Currently the return value is defined as
> Functions snprintf() and vsnprintf() don't write more than count bytes, including the terminating null byte ('\0'). If the output was truncated due to this limit, the return value is count, which indicates any truncation.

The "hello, world" produces 12 chars plus the trailing \0. 12 would normally be returned.
The count size of 10 in the example above is the limit. As truncation occurs, 10 is returned and "hello, wo"\0 is written.

I think, it's okay.

Snaipe, 2018-05-03:
This interpretation is somewhat incorrect. According to the [man page](https://linux.die.net/man/3/vsnprintf) (emphasis mine, also see 7.19.6.5.3 from the C99 spec):

> The functions snprintf() and vsnprintf() do not write more than *size* bytes (including the terminating null byte ('\0')).  **If the output was truncated due to this limit, then the return value is the number of characters (excluding the terminating null byte) which would have been written to the final string if enough space had been available**.  Thus, a return value of *size* or more means that the output was truncated.

Which means that in this case, the return value should be `12`, not `10`. This behaviour is actually very important for `vsnprintf`, as it allows one to call `vsnprintf(NULL, 0, fmt, vl)` to determine the appropriate size of a buffer suitable for formatting.

mpaland, 2018-05-05:
I see your point here and I know that the current behavior is **not conform to specification**.

In the first way I thought that the `printf` return value doesn't care most people (including me) at all.
To get the length of a formatted string or to format a string itself, there are many libs around, like fmt.

I tried to find a balance between code size, complexity and supported features. My goal was not to write a 100% feature complete 1:1 plugin for the original `printf`.

Anyway, I have an idea to refactor the output a little bit to get a spec conform return value, but this has no high prio.
I leave this issue open for tracking.

Snaipe, 2018-05-07:
AFAIK `fmt` requires a C++ compiler; and other solutions that are C-only aren't quite as complete.

Reading the code, I see the rationale behind the current behaviour, and I understand why it wouldn't be that high priority. I'm mostly interested in this issue because I'm looking for a replacement for the printf function family on Windows, because their implementation is horrible to work with when manipulating UTF-8 strings (and even the mingw implementation has a weird locale-aware behaviour that's quite inconsistent when formating characters not in the "preferred code page").

I could submit a PR when I get the time, if you don't mind.

mpaland, 2018-05-07:
Hello @Snaipe , I don't mind at all!

My idea for refactoring this is to have a central (string) buffer write routine which checks for buffer maxlen and stop writing accordingly. This could eliminate a lot of `(idx < maxlen)` and simply the code.
I'll check this out.

Leandros, 2018-05-08:
Awesome! Looks great! One minor nitpick, though. You could move the `NULL` check out of the `_out_buffer` routine, and simply pass in an empty function when the output buffer is null. That would remove the branches in this case and slightly improve performance.

mpaland, 2018-05-08:
@Leandros , pretty cool idea! Implemented. ✔️

commit 33f10559 (2018-05-11): refactor(printf): usage of (buffer) output routine, fixes return value

Buffer for printf is no longer needed, chars are directly written now.
Return value is specification conform now.
Fixes #11
closed 2018-05-11 by mpaland