mpaland/printf #13: Scawy big unkown type UwU

MikuChan03, 2018-05-06:
Hi!!

Using your kawaii pwintf impwementation,
the compiler gets all funny about some silly size_t.

printf.h:78:29: error: unknown type name 'size_t'
 int  snprintf(char* buffer, size_t count, const char* format, ...);
                             ^~~~~~

It even dwew a line for us owo
I did some wesearch and it said, there's no sizy-wizy_t in stdarg.h.
Instead, I moved stddef.h fwom pwintf.c to pwintf.h, because
stddef.h is like totally the hood of size_t.

This fixie wixies the pwoblem
xD *starts twerking*

mpaland, 2018-05-06:
You are right.
This is a regression introduced by cd4481a6f29893fb0fff0c0ff0ff9c8f152af7dc . Sorry for that!
Neither the test suit nor my projects suffered from this problem cause <stddef.h> war included indirectly.

closed 2018-05-06 by mpaland

Y32bit, 2021-02-17:
what

commit 814bbf6f (2018-05-06): fix(printf): move <stddef.h> to header

Fixes #13