mpaland/printf #15: Scawy big no functionality at all UwU

MikuChan03, 2018-05-12:
Senpai,

your little kawaii pwintf impwementation is getting out of contwol.
Pwease have a look at this:

The pwintf function:
int printf(const char* format, ...)
{
  va_list va;
  va_start(va, format);
  const int ret = _vsnprintf(_out_char, NULL, (size_t)-1, format, va);
  va_end(va);
  return ret;
}

The _vsnpwintf function:
static int _vsnprintf(out_fct_type out, char* buffer, const size_t maxlen, const char* format, va_list va)
{
  unsigned int flags, width, precision, n;
  size_t idx = 0U;

  if (!buffer) {
    // use null output function
    out = _out_null;
  }
...

Since we pass null fow the buffer, the output function is being set to _out_null,
which doesn't do anything. Pwintf has ceased functioning entirely.

Oh no!

mpaland, 2018-05-13:
Sorry for inconvenience. That wasn't covered by any test case. Fixed now. Thanx!

closed 2018-05-13 by mpaland

BRA1L0R, 2020-07-10:
What the fuck

Fark1, 2021-02-16:
lmao  what the fuck

TroyDota, 2021-02-16:
hello friends.

SpheeresX, 2021-02-16:
hello from reddit

ReardGjoni, 2021-02-16:
Did it get fixie wixied?

larbikhounti, 2021-02-16:
😂

regnaio, 2021-02-16:
(づ｡◕‿‿◕｡)づ

commit d12e52be (2018-05-13): fix(printf): pass dummy buffer for printf to _vsnprintf instead of NULL

Fixes #15