mpaland/printf #16: Rename functions and use macros to avoid conflicts with STD

reliasn, 2018-05-28:
In ``printf.h`` you would have something like this:
```c
void x_putchar(char character);
int x_printf(const char* format, ...);
int x_sprintf(char* buffer, const char* format, ...);
int x_snprintf(char* buffer, size_t count, const char* format, ...);
int x_vsnprintf(char* buffer, size_t count, const char* format, va_list va);
int x_fctprintf(void (*out)(char character), const char* format, ...);
```
Along with a set of macros:
```c
#define vsnprintf x_vsnprintf
#define snprintf x_snprintf
...
```
This is what [tinyprintf](https://github.com/cjlano/tinyprintf/blob/master/tinyprintf.h#L166-L172) does and it is very convenient because it avoids conflicts with the STD printf functions. 
[image]

I'm actually coming from [tinyprintf](https://github.com/cjlano/tinyprintf) because it lacks float support and doesn't seem to be updated often.

mpaland, 2018-05-29:
@reliasn thanx for your feedback!
I know `tinyprintf`, receiving no more updates may not be a bad thing, if the code is mature and bug free. :smiley:  
`tinyprintf` just doesn't pass the test suite very well.

Anyway, the old discussion about already defined functions in the standard lib. I'm really well aware of that problem. Normally it's not a problem when you do not link against the standard, but if you need due to other functions, things become difficult.

I tried to avoid using function redeclaring defines, because not all compilers fully support variadic macrcos (even if they should) by now.

A possible solution might be a combination like that:
In printf.c:
```c
#ifdef OVERRIDE_LIBC
int x_printf(const char    )
#else
int printf(const char    )
#endif
{
  // impl
}
```
And in printf.h:
```c
#ifdef OVERRIDE_LIBC
#define printf( , ...)  x_printf(const char  , ...)
#else
int printf(const char  , ...);
#endif
```

mpaland, 2018-06-05:
In the case above a variadic macro is not necessarily needed.
Solution is in work.

DBJDBJ, 2018-09-01:
Hi, excellent lib, clean and sober code, really good this is. Now.
I am wondering if you could deliver cpp header only version? 
Yes, I know that would be a "point-of-no-return" to C .., And of course we are all crazy busy, I am aware of that too :)  Ok let me see what I can do ...

DBJDBJ, 2018-09-01:
# [printf++](https://github.com/DBJDBJ/printf_plusplus)

commit 54dfd185 (2018-11-02): feat(printf): added PRINTF_OVERRIDE_LIBC support

Fixes #16
closed 2018-11-02 by mpaland