mpaland/printf #19: printf appends null terminator to formatted output

acesme, 2018-06-21:
Hello,
Thank you for the fantastic work.
Should printf append a null terminator at the end of the formatted output? The standard seems to be unclear about this. The only statement I could find in the c11 standard regarding this is "The fprintf function returns when the end of the format string is encountered." (according to the standard, printf is equivalent to fprintf except for the first argument).

mpaland, 2018-06-21:
Hello @acesme ,
thank you for your comment!
As a matter of fact, it's a really good question. I would say (or suggest) it shouldn't.
In the moment, the low level `_putchar()` function is responsible for the `printf()` output, so any \0 could be filtered out easily.
Normally the plain native `printf()`is used to output strings to the screen or a terminal and a terminator isn't needed or even unwanted.
Have to check it... Any other opinions/suggestions here?

acesme, 2018-06-21:
Yes, it is simple to filter out the final `'\0'` in the `_putchar()` implementation. 
It is also possible to fix this on 653rd row of printf.c by checking if `out == _out_char` and opting to not append the `'\0'` (or something like that).
I have no other suggestions that I can think of for now.

mpaland, 2018-06-21:
I think, it's better to "fix" this in _out_char() itself, so other output functions won't get any penalty, something like
```C
static inline void _out_char(char character, char* buffer, size_t idx, size_t maxlen)
{
  (void)buffer; (void)idx; (void)maxlen;
  if (character) {
    _putchar(character);
  }
}
```

closed 2018-06-24 by mpaland

mpaland, 2018-06-24:
I think, suppressing the terminating `\0` is conform to the standard.
@acesme, thanx for bringing this to attention. Fixed.

commit 810cb2b2 (2018-06-24): fix(printf): suppress terminating \0 in printf() output

Closes #19