mpaland/printf #22: Incorrect test case

akbertram, 2018-08-21:
The "float" test suite contains the following assertions:

    test::sprintf(buffer, "%.9f", 42.8952);
    REQUIRE(!strcmp(buffer, "42.895200000"));

    test::sprintf(buffer, "%.10f", 42.895223);
    REQUIRE(!strcmp(buffer, "42.895223000"));

The first assertion looks correct, but the second assertion appears to be wrong: the precision is set to 10 and there are only 9 digits after the decimal point. 

I've checked this with GLIBC 2.23-0ubuntu10 and indeed, 10 digits appear after the decimal point.

mpaland, 2018-08-21:
Hello Alex,
thanks a lot for your investigation and bringing this to attention.

I know that the float conversion ain't really pretty and even lacks exponential support. It's still a kind of workaround and perhaps I get some help from @ckormanyos in the wintertime to improve the floating-point conversion.

So, yeah, the testcase you mentioned is actually a valid one for *this* printf implementation, because the precision of 10 is silently limited to 9 digits, so having "%.10f" returns 9 digits after the point. That is expected here - but I agree, this is not the correct way and may be considered as a bug.
And I should mention this explicit in the readme.

Anyway, I fix this and return the given number of precision digits while returning zeros after the 9th digit.
The testcase above will become
```c
  test::sprintf(buffer, "%.12f", 42.89522312345678);
  REQUIRE(!strcmp(buffer, "42.895223123000"));
```
then.
This is not perfect as well, but better than returning a wrong precision.

akbertram, 2018-08-21:
Thanks for your response, and for the test suite! The tests have proven very useful in verifying our Java implementation of sprintf()!

commit e6b5331a (2018-08-21): fix(printf): fix floating point precision limit

Return the correct count of precision digits now. Fixes #22
closed 2018-08-21 by mpaland