mpaland/printf #25: Passing negative precision is broken

wm4, 2018-09-13:
An indirect precision (as in ``.*``) is passed as a signed int. But this code interprets it as unsigned. C11 even specifies that ``negative precision is taken as if the precision were omitted``.

Test case: ``printf("%.*d", -1, 1)`` should output ``1``.

mpaland, 2018-09-14:
Good catch!
This case is indeed not implemented. 😞

closed 2018-09-14 by mpaland

ledvinap, 2019-04-18:
There is still subtle bug: `sprintf("%.*d", -1, 0);`  should produce `"0"` (precision omitted), not "" (precision specified and zero)

wm4, 2019-04-18:
Over the time, I found some more subtle issues that (potentially) mismatch with standards or generally expected behavior. They probably don't matter for embedded use at all. Anyway, I used the test suite from this project as base, ported it to C, and added some more test cases, so if you want: https://github.com/wm4/libinsanity/blob/master/tests/printf_test.c
(That project doesn't target embedded use at all.)

ledvinap, 2019-04-18:
@wm4 thanks for the link.

I'm now playing with printf, it should be easy to fix and I'll try other cases

commit 6dae1687 (2018-09-14): fix(printf): fix negative argument precision

Fixes #25