mpaland/printf #26: Problems with zero precision

wm4, 2018-09-13:
There are some problems with a precision of 0 (an annoying corner case).

The following crash:
- ``printf("% .0d", 0);`` (should output `` ``)
- ``printf("%+.0d", 0);`` (should output ``+``)

This is caused by decrementing a 0 ``len`` here: https://github.com/mpaland/printf/blob/master/printf.c#L185

``printf("%#.0x", 0);`` is apparently supposed to output an empty string, not ``0x``. The octal case is correct, though. It seems this was clarified in C11.

mpaland, 2018-09-14:
Wow! This is really serious and leads to out of bounds buffer access!
Wondering that coverity didn't catch it.

closed 2018-09-14 by mpaland

eyalroz, 2021-07-21:
The fix for this bug may have causes #109 .

commit 7075d314 (2018-09-14): fix(printf): zero precision and zero value hash problem

Fixes #26