mpaland/printf #30: Floats : out of range

NotSqrt, 2018-09-19:
Hi,

What is missing exactly to support `double` numbers ?

I see the message in the tests : https://github.com/mpaland/printf/blob/21a282a2a427cb33d12d1d31ae8aee14d9bbb8c3/test/test_suite.cpp#L1016-L1017

Maybe for the moment, a message can be added to the caveats in the README : 
> Double precision is not supported and will result in an empty string.

Thanks !

NotSqrt, 2018-09-19:
Is that due to this ?
https://github.com/mpaland/printf/blob/c235b5cd6eb541f87de4ff76487ca39c20153f5b/printf.c#L322-L326

This is the only missing feature I found compared to `xprintfc` (which is a bit smaller but not compliant with the standard, and not really tested). 
They use this code to handle floats : https://github.com/MarioViara/xprintfc/blob/8f07c7c0d76370cdb44dbaee81f1f4201b1e7561/src/xformatc.c#L773 
and their tests use `printf("Floating > 32 bit %f",pow(2.0,32.0)+1.0);`

ckormanyos, 2018-11-20:
Have you seen this? [https://github.com/ulfjack/ryu](url)
It seems like it might still be experimental for C, but looks very promising...

mpaland, 2019-03-26:
This is closed by 4.0.0

closed 2019-03-26 by mpaland