mpaland/printf #37: Warning: comparing floating point with == or != is unsafe [-Wfloat-equal]

HarjitSi, 2018-12-23:
Compiling with GCC using -Wfloat-equal turned on, I get the following warnings:
printf.c:346:18: warning: comparing floating point with == or != is unsafe [-Wfloat-equal]
printf.c:363:20: warning: comparing floating point with == or != is unsafe [-Wfloat-equal]

The code does has a compare to 0.5

Any ideas on how to resolve this?
Thank you for creating this wonderful function!

mpaland, 2018-12-23:
Hello @HarjitSi,

thanks a lot for bringing this problem to attention.
I'm aware of this "bug" and I don't like it either. Comparing floats to discrete numbers is never a sure/good thing.
The whole `ftoa` routine needs further testing and minor refactoring. This is still on the todo list.
Anyway, this is just a warning but if you need to be warning free, a construct of greater/smaller comparisons might help - as a quick fix.

mpaland, 2018-12-24:
@vgrudenic , thanks a lot for your suggestion.
Is `double` really and always 8 bytes on any platform?
I have to investigate this a little further...

Merry christmas! 🎄

vgrudenic, 2018-12-24:
Hi, merry christmas to you too! 🎄😃

No, I believe the C standard says [nothing about the size](https://stackoverflow.com/q/25524355/69809), although IEEE 754 / IEC 60559 specifies float and double as 32 and 64-bit. 

Perhaps my comment was unnecessarily complicated, and the most reasonable option would be to suppress the warning in GCC? This should leave it to compiler to produce the correct code for a given platform, regardless of the size of floats:

```C
// doesn't have to be a separate method
static inline bool double_are_equal(double a, double b)
{

// temporarily suppress "comparing float with == is unsafe"
// (we really want to check exactly against 0.5)
#ifdef __GNUC__
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wfloat-equal"
#endif

    return a == b;

#ifdef __GNUC__
#pragma GCC diagnostic pop
#endif

}
```

This will probably result in the

commit 80b42fef (2019-01-26): fix(printf): fixed support of NaN

(hot) fixes #37
closed 2019-01-26 by mpaland