mpaland/printf #38: Why did you rename the functions to have an appended _?

phillipjohnston, 2019-01-15:
Just curious about the motivation behind the name change (to have functions appended with an underscore, and then redefined without the underscore). 

```
#define printf printf_
int printf_(const char* format, ...);
```

This caused some problems with my library when I updated. I have a libc which I use for embedded systems, but on host machines I link against the standard printf. With the new naming convention, I can no longer keep the printf.h header included and link against the std library - `printf_` is undefined, since the library defines `printf`.

mpaland, 2019-01-15:
Hello @phillipjohnston,
sorry for any inconvenience. Motivation of the trailing underscore is to avoid any trouble while linking this module.
Some toolchains use/link the standard libc which defines the common `printf()` functions. In this case the following scenarios might happen (depending on the used linker):
1. The linker takes the printf module and ignores the libc functions. 👍 
2. The linker takes/uses the `printf()` function it finds first. This might be the (wrong) libc function. 👎 
3. The linker gives a duplicate defined symbol error. 👎 

I've seen all variants. To prevent case 2 and 3 `printf` is defined to `printf_` and `printf_()` has no duplicate definition.
So any libc `printf()` function is unused and optimized away by the linker.

When you *use* printf for creation of your libc, that might be a problem indeed, because `printf()` (without underscore) will not be defined.

I can create a compiler switch, to define printf_ as printf...

phillipjohnston, 2019-01-15:
Thanks, that makes sense. I was relying on behavior 2 implicitly. Let me see if I can unwind everything on my end as well to where the compiler switch would be unnecessary.

mpaland, 2019-03-26:
I close this issue so far, as there are no complains about the current implementation using defines and underscores.

closed 2019-03-26 by mpaland