mpaland/printf #52: Discard round-to-even code?

ledvinap, 2019-04-20:
Round-to-even in `%f` is not working correctly, maybe it would be best to simply discard it completely? Opinions?

```
  unsigned long whole = (unsigned long)value;
  double tmp = (value - whole) * pow10[prec];
  unsigned long frac = (unsigned long)tmp;
  diff = tmp - frac;
```
The problem is that `tmp` value is multiplied by `pow10`. 

With `%.1f`, 0.95:
0.95 is represented as 0.94999999999999996 in binary. This should be rounded down.
But `(0.95 - 0) * 10` is 9.5, exactly representable in binary and tie-decision code takes place, rounding to even and outputting `"1.0"`.

It may be possible to fix this double-rounding (at least in some cases), but IMO it's not worth the complexity. It may be simpler to just remove  the tie-handling code and live with it

ledvinap, 2019-04-20:
One edge case is `%.0f` - current rounding computation is correct and .5 is exactly representable in binary, so rounding should apply.

mpaland, 2026-10-31:
You're right, the double rounding can't be fixed in a cheap way: `0.95` is slightly below 0.95 in binary, but `0.95 * 10` is exactly 9.5 and is rounded half to even.
I decided to keep the tie handling, because it gives the expected results for exact ties like `%.0f` of 2.5 or `%.1f` of 0.25, and the carry bug at a tie (#51) is fixed in v4.1.0.

The behaviour is now documented in the README under *Caveats*, with exactly the `%.1f` of 0.95 example, and covered by tests.
A correctly rounded conversion would need big-number arithmetic, which is out of scope for this library.

closed 2026-10-31 by mpaland