mpaland/printf #54: floating point zero handling

ledvinap, 2019-04-23:
When printing "%e" or "%g" with zero as argument, 0e-308 is assumed instead of zero

```
printf("%e", 0):
expected: "0.000000e+00"
got: "0.000000e-308"

printf("%g", 0):
expected: "0"
got: "0.000000e-308"

eyalroz, 2021-08-02:
Well, the problem is in `_etoa`, which has a line which estimates the base-10 exponent of the value to print using its base-2 exponent:
```
int expval = (int)(0.1760912590558 + exp2 * 0.301029995663981 + (conv.F - 1.5) * 0.289529654602168);
```
This uses a linear approximation of the ln(x) function at x = 1.5 . This is itself a bit fishy, but I suppose we could live with it... for normal floats. With 0.0, it is [sub-normal](https://en.wikipedia.org/wiki/Denormal_number) - the most extreme case of sub-normality.

We have two options on how to handle this:

1. Fix denormal handling generally by counting the leading 0 bits of the mantissa when the exponent is 0
2. Just special-case 0 and ignore handling of other denormals.

I intend to go with option (2.) for now, and open a separate bug for (1.) - unless there are any objections.

eyalroz, 2021-08-02:
So, fixed on the development branch of [my fork](https://github.com/eyalroz/printf/).

closed 2026-10-31 by mpaland