mpaland/printf #62: Another project using this library :)

phillipjohnston, 2019-10-09:
I've added a wrapper around your library for projects using the Arduino SDK:

https://github.com/embeddedartistry/arduino-printf

Still my favorite printf library!

salkinium, 2020-01-01:
[modm.io](https://modm.io/reference/module/modm-io/#using-printf) is also using this library, albeit with some [hacky hacks](https://github.com/modm-io/modm/blob/develop/src/modm/io/iostream_printf.cpp.in) to disable the native floating point functions on AVR and use the optimized (fixed options) routines from avr-libc.

At least for our use-case it would've been more advantageous to have a more modular internal API, (we use the internal "static" functions `_ntoa_long_long`, `_etoa` etc too), to allow for a more custom printf build. However, that may not be a universal requirement 😛.

vhelin, 2020-02-12:
I just integrated this library to https://github.com/vhelin/wla-dx/issues/273 for the implementation of snprintf() - I hope this is ok. I had to modify the code a little bit to make this library ANSI C90 compatible...

closed 2026-11-02 by mpaland