mpaland/printf #63: printf("%s", NULL) - crash

mad-rain, 2019-10-10:
Hi!

Some placeholder should be used for NULL %s argument to prevent printf from crash:

```
      case 's' : {
        const char* p = va_arg(va, char*);
        if (!p) p = "(null)"; // please add this
```

noomio, 2019-10-17:
I just came here to add just that!

        const char* p = va_arg(va, char*);
        p = (p == NULL) ? "(null)" : p;

dev-zzo, 2020-04-22:
This should be an optional behaviour as in many embedded situations address zero is a perfectly valid one.

eyalroz, 2021-06-30:
@dev-zzo : What about `%p`, then? Should that also not have an option to print 0 as 0 rather than (nil) on such platforms?

eyalroz, 2021-06-30:
So, this issue is now fixed on my [fork](https://github.com/eyalroz/printf/). @dev-zzo , @noomio , @mpaland  Please have a look and voice your opinion about the behavior with null/0 pointers.

Also, @mpaland : Can you explain the code's "partiality" to the use of _out_rev? That is, what's the benefit of printing characters backwards like that? I used it too in the changes I made for this issue, but I was just following the pattern.

closed 2026-10-31 by mpaland