mpaland/printf #66: proposal: additional prefix to avoid name collision

diggit, 2019-11-22:
I propose to add prefix to all exposed functions using eg. macro. This is to avoid possible name collision with stdio and other libraries when used.

What do you think?

mpaland, 2020-01-01:
All exposed functions end with `_`, and there are defines which define the function name, e.g.
``` c
#define printf printf_
int printf_(const char* format, ...);
```
So IMHO there won't be a function name collision.

ryanwinter, 2020-02-14:
I noticed that <cstdio> it does an:
`#undef printf`

which results in a linker error when it tries to pull the global printf into the std namespace:
`cstdio:127:11: error: '::printf' has not been declared`

I don't understand why it can't find the original printf defined in stdio.h, but the <cstdio> is included by a 3rd party library.

eyalroz, 2021-08-03:
On my [fork](https://github.com/eyalroz/printf), what I've done so far is make the aliasing of the standard-library function be disabled by default, with a CMake-controlled `#define` allowing it to happen if necessary.