mpaland/printf #68: Caveat: requires remainder, divide, and multiply instructions or software implementations

jamolnng, 2020-01-09:
This came from this reddit thread https://web.archive.org/web/20200109232034/https://www.reddit.com/r/RISCV/comments/eme6qb/gcc_error_undefined_reference_to_umodsi3/

If the computer architecture does not include remainder, divide, or multiply instructions or software implementations this does not compile. Such instance would be a processor that only implements rv32I instruction set as seen in the reddit thread.

jamolnng, 2020-01-10:
This implies that you aren't linking the libgcc that the compiler supplied, which should be done but doesn't have to be

mpaland, 2020-01-10:
> 
> This implies that you aren't linking the libgcc that the compiler supplied, which should be done but doesn't have to be

Is this a printf question?
Normally the project defines what need to be linked.
printf has nothing to do with underlaying libs nor it needs any special ones.

closed 2020-01-10 by mpaland