mpaland/printf #71: request: sprintf() comma compatibility

simonhf, 2020-02-03:
```
#include <stdio.h>

// gcc           -O0 -o mpaland mpaland.c && ./mpaland
// gcc -DMPALAND -O0 -o mpaland mpaland.c && ./mpaland

#ifdef MPALAND
void _putchar(char character)
{
    putchar(character);
}

#include "printf/printf.c"
#endif

void main(void) {
    printf("- main() starting\n");
    printf("- value with comma: %'d\n", 1234);
    printf("- main() ending\n");
}
```

Run without mpaland printf:

```
$ gcc           -O0 -o mpaland mpaland.c && ./mpaland
- main() starting
- value with comma: 1234
- main() ending
```

Run with mpaland printf:

```
$ gcc -DMPALAND -O0 -o mpaland mpaland.c && ./mpaland
- main() starting
- value with comma: 'd
- main() ending
```

It would be great if mpaland / printf could be compatible with the sprintf() format comma syntax, even if commas are not printed due to the locale (as with the above example via glibc). It would be even better if mpaland / printf implemented comma output too, although obviously not by manipulating the locale...

[1] https://stackoverflow.com/questions/1449805/how-to-format-a-number-from-1123456789-to-1-123-456-789-in-c

simonhf, 2020-02-03:
Also, I had to add this #ifdef in this other example of using mpaload / printf() otherwise it seg fault. Presumably this is linked to not understanding the comma syntax and therefore the number of parameters somehow are not dealt with correctly?

[1] https://gist.github.com/simonhf/2a7b7eb98d2a10c549e8cc858bbefd53#file-svnprintf-c-L58

mpaland, 2020-02-16:
Hi Simon,
you mean the apostrophe flag. Yeah, that's a little difficult due to system locale.
Number formating is not finally done an there are some good PRs here. The system locale (comma or dot) would/could be a config define, I won't like a querying that.
I have a look on your PR.

simonhf, 2020-02-19:
Hi Marco, Yep, I understand. I like the idea of the config define. My PR does not implement the formatting, but is a pre-cursor step allowing the apostrophe to be present in the format string even though it has no effect. Without the PR then the apostrophe messes up the format string and/or causes a seg fault...

closed 2026-11-11 by mpaland