mpaland/printf #75: Format "%.4g" not working as expected

fbeaulier, 2020-03-10:
I tested `printf("%.4g",  (double)4);` and it displays 4.000 instead of just 4
I have checked to be sure, the standard libc with gcc on Linux displays 4

essele, 2021-02-23:
Hi ... I've just found this issue also .. just running through a few tests and actually %g has quite a few issues...

In the below STD is the normal printf (this is a Mac, but it's the same on Linux), and NEW is this printf.

Lee.

`Using [%g]`
`[%g        ]  STD (0              )  NEW (0.000000e-308  )`
`[%g        ]  STD (1e-08          )  NEW (1.000000e-08   )`
`[%g        ]  STD (1e-07          )  NEW (1.000000e-07   )`
`[%g        ]  STD (0.0001         )  NEW (0.000100000    )`
`[%g        ]  STD (1.45679        )  NEW (1.45679        )`
`[%g        ]  STD (100            )  NEW (100.0000       )`
`[%g        ]  STD (100.001        )  NEW (100.001        )`
`[%g        ]  STD (100            )  NEW (100.000        )`
`[%g        ]  STD (112.295        )  NEW (112.295        )`
`[%g        ]  STD (4.84848e+08    )  NEW (4.848484e+08   )`
`[%g        ]  STD (-100           )  NEW (-100.0000      )`
`[%g        ]  STD (-0.0001        )  NEW (-0.000100000   )`
`[%g        ]  STD (-50            )  NEW (-50.0000       )`
`[%g        ]  STD (-5000          )  NEW (-5000.00       )`
`[%g        ]  STD (-1e-08         )  NEW (-1.000000e-08  )`
`[%g        ]  STD (-4.48488e+08   )  NEW (-4.484885e+08  )`
`Using [%.4g]`
`[%.4g      ]  STD (0              )  NEW (0.000e-308     )`
`[%.4g      ]  STD (1e-08          )  NEW (1.000e-08      )`
`[%.4g      ]  STD (1e-07          )  NEW (1.000e-07      )`
`[%.4g      ]  STD (0.0001         )  NEW (0.0001000      )`
`[%.4g      ]  STD (1.457          )  NEW (1.457          )`
`[%.4g      ]  STD (100            )  NEW (100.00         )`
`[%.4g      ]  STD (100            )  NEW (100.0          )`
`[%.4g      ]  STD (100            )  NEW (100.0          )`
`[%.4g      ]  STD (112.3          )  NEW (112.3          )`
`[%.4g      ]  STD (4.848e+08      )  NEW (4.848e+08      )`
`[%.4g      ]  STD (-100           )  NEW (-100.00        )`
`[%.4g      ]  STD (-0.0001        )  NEW (-0.0001000     )`
`[%.4g      ]  STD (-50            )  NEW (-50.00         )`
`[%.4g      ]  STD (-5000          )  NEW (-5000          )`
`[%.4g      ]  STD (-1e-08         )  NEW (-1.000e-08     )`
`[%.4g      ]  STD (-4.485e+08     )  NEW (-4.485e+08     )`
`Using [%4g]`
`[%4g       ]  STD (   0           )  NEW (0.000000e-308  )`
`[%4g       ]  STD (1e-08          )  NEW (1.000000e-08   )`
`[%4g       ]  STD (1e-07          )  NEW (1.000000e-07   )`
`[%4g       ]  STD (0.0001         )  NEW (0.000100000    )`
`[%4g       ]  STD (1.45679        )  NEW (1.45679        )`
`[%4g       ]  STD ( 100           )  NEW (100.0000       )`
`[%4g       ]  STD (100.001        )  NEW (100.001        )`
`[%4g       ]  STD ( 100           )  NEW (100.000        )`
`[%4g       ]  STD (112.295        )  NEW (112.295        )`
`[%4g       ]  STD (4.84848e+08    )  NEW (4.848484e+08   )`
`[%4g       ]  STD (-100           )  NEW (-100.0000      )`
`[%4g       ]  STD (-0.0001        )  NEW (-0.000100000   )`
`[%4g       ]  STD ( -50           )  NEW (-50.0000       )`
`[%4g       ]  STD (-5000          )  NEW (-5000.00       )`
`[%4g       ]  STD (-1e-08         )  NEW (-1.000000e-08  )`
`[%4g       ]  STD (-4.48488e+08   )  NEW (-4.484885e+08  )`
`Using [%4.4g]`
`[%4.4g     ]  STD (   0           )  NEW (0.000e-308     )`
`[%4.4g     ]  STD (1e-08          )  NEW (1.000e-08      )`
`[%4.4g     ]  STD (1e-07          )  NEW (1.000e-07      )`
`[%4.4g     ]  STD (0.0001         )  NEW (0.0001000      )`
`[%4.4g     ]  STD (1.457          )  NEW (1.457          )`
`[%4.4g     ]  STD ( 100           )  NEW (100.00         )`
`[%4.4g     ]  STD ( 100           )  NEW (100.0          )`
`[%4.4g     ]  STD ( 100           )  NEW (100.0          )`
`[%4.4g     ]  STD (112.3          )  NEW (112.3          )`
`[%4.4g     ]  STD (4.848e+08      )  NEW (4.848e+08      )`
`[%4.4g     ]  STD (-100           )  NEW (-100.00        )`
`[%4.4g     ]  STD (-0.0001        )  NEW (-0.0001000     )`
`[%4.4g     ]  STD ( -50           )  NEW (-50.00         )`
`[%4.4g     ]  STD (-5000          )  NEW (-5000          )`
`[%4.4g     ]  STD (-1e-08         )  NEW (-1.000e-08     )`
`[%4.4g     ]  STD (-4.485e+08     )  NEW (-4.485e+08     )`
`Using [%.14g]`
`[%.14g     ]  STD (0              )  NEW (0.0000000000000)`
`[%.14g     ]  STD (1e-08          )  NEW (0.9999999890000)`
`[%.14g     ]  STD (1e-07          )  NEW (0.9999999990000)`
`[%.14g     ]  STD (0.0001         )  NEW (0.0001000000000)`
`[%.14g     ]  STD (1.456789       )  NEW (1.4567890000000)`
`[%.14g     ]  STD (100            )  NEW (100.00000000000)`
`[%.14g     ]  STD (100.001        )  NEW (100.00100000000)`
`[%.14g     ]  STD (100.000004     )  NEW (100.00000400000)`
`[%.14g     ]  STD (112.294757     )  NEW (112.29475700000)`
`[%.14g     ]  STD (484848444      )  NEW (4.8484844400000)`
`[%.14g     ]  STD (-100           )  NEW (-100.0000000000)`
`[%.14g     ]  STD (-0.0001        )  NEW (-0.000100000000)`
`[%.14g     ]  STD (-50            )  NEW (-50.00000000000)`
`[%.14g     ]  STD (-5000          )  NEW (-5000.000000000)`
`[%.14g     ]  STD (-1e-08         )  NEW (-0.999999989000)`
`[%.14g     ]  STD (-448488484     )  NEW (-4.484884840000)`

eyalroz, 2021-07-14:
@essele : Does your repository address all of these?

eyalroz, 2021-07-30:
@fbeaulier : In fact, it's even worse than that. The current test suite has tests of the `%g` specifier with _incorrect_ expected results - specifically in these cases:
```
  test::sprintf(buffer, "%.6G", 12345.);
  REQUIRE(!strcmp(buffer, "12345.0"));
```
actually, it should be `12345`, no decimal point and no trailing zero. And:
```
  test::sprintf(buffer, "%0-15.3g", -42.);
  REQUIRE(!strcmp(buffer, "-42.0          "));
```
actually, it should be `42`, no decimal point and no trailing zero.

However - I'm about to resolve this on my fork.

closed 2026-10-31 by mpaland