mpaland/printf #78: printf %g float passing on ARM crash

chuckb, 2020-05-09:
I am compiling with gcc for ARM target...raspberry Pi zero. When I attempt to pass a float, I get a crash. CPU start up code enables the VFP with:
```
    // Enable VFP ------------------------------------------------------------

    // r1 = Access Control Register
    MRC p15, #0, r1, c1, c0, #2
    // enable full access for p10,11
    ORR r1, r1, #(0xf << 20)
    // Access Control Register = r1
    MCR p15, #0, r1, c1, c0, #2
    MOV r1, #0
    // flush prefetch buffer because of FMXR below
    MCR p15, #0, r1, c7, c5, #4
    // and CP 10 & 11 were only just enabled
    // Enable VFP itself
    MOV r0,#0x40000000
    // FPEXC = r0
    FMXR FPEXC, r0
```

My compile statement is like the following:
```
arm-none-eabi-gcc  -O2 -DRPI0 -Wall -fno-builtin -nostartfiles -fno-exceptions -fno-unwind-tables -mfpu=vfp -mfloat-abi=hard -mcpu=arm1176jzf-s -I"/Users/chuck_benedict/Projects/testHaikuVM/." -I"../../HaikuVM/src"  -I"../../HaikuVM/src/utility" -c -o "HaikuVM/src/heap.o" "../../HaikuVM/src/heap.c"
Finished building: ../../HaikuVM/src/heap.c
```

My link statement is:
```
arm-none-eabi-g++ -Wl,-T,/Users/chuck_benedict/Projects/haikuVM/haikuVM/src/lib/ldscripts/rpi.x,-Map,Main.map -nostartfiles -mfpu=vfp -mfloat-abi=hard -mcpu=arm1176jzf-s -o"Main.elf" -lm  ./HaikuVM/src/platforms/pi/printf.o ./HaikuVM/src/platforms/pi/rpi-gpio.o ./HaikuVM/src/platforms/pi/rpi-systimer.o ./HaikuVM/src/platforms/pi/rpi-armtimer.o ./HaikuVM/src/platforms/pi/rpi-aux.o ./HaikuVM/src/platforms/pi/armc-stubs.o ./HaikuVM/src/platforms/pi/rpi-mailbox.o ./HaikuVM/src/platforms/pi/rpi-mailbox-interface.o ./HaikuVM/src/platforms/pi/rpi-interrupts.o  ./HaikuVM/src/platforms/platform_AVR.o ./HaikuVM/src/platforms/platform_TEENSYDUINO.o ./HaikuVM/src/platforms/platform_RCX.o ./HaikuVM/src/platforms/platform_RPI.o ./HaikuVM/src/platforms/platform_WIN.o ./HaikuVM/src/platforms/platform_UNIX.o  ./HaikuVM/src/gcVariants/incremental.o ./HaikuVM/src/gcVariants/conservative.o ./HaikuVM/src/gcVariants/no.o  ./HaikuVM/src/utility/haikuJNI.o ./HaikuVM/src/utility/haikuConfig.o ./HaikuVM/src/utility/nativeCFunctions.o  ./HaikuVM/src/new4avrgcc.o ./HaikuVM/src/JVM.o ./HaikuVM/src/heap.o ./HaikuVM/src/HaikuAppMain.o ./HaikuVM/src/haikuJavaNatives.o ./HaikuVM/src/haikuJ2C.o ./HaikuVM/src/halfprecision.o ./HaikuVM/src/romspace.o    -lc
```

Can you help me with the magic incantation required to get this to work?

AshkanAnousheh, 2020-09-10:
hi @chuckb ,
change your Floating Point setting to "Software Implementation" could solve your problem for now.

@mpaland thanks for your great library, please fix the bug.

ledvinap, 2020-09-10:
Is it possible to get context of crash?
`gdb` backtrace + other printed info could help ...

AshkanAnousheh, 2020-09-10:
This image shows Cortex-M4 processor status after vsnprintf function executed.
[image]

Program walk-through functions:
vsnprintf_()  -> _vsnprintf()  -> signal handler called at 0xfffffff9 -> stuck at WWDG_IRQHandler()

This error log may have useful information:
Develop Log System needs property 'FPU' fixed. Fixing from 'com.atollic.truestudio.common_options.target.fpucore.DUMMY' to 'com.atollic.truestudio.common_options.target.fpucore.fpv4-sp-d16'.

ledvinap, 2020-09-12:
This is a bit strange - Window Watchdog IRQ is raised, either in place unrelated to FPU or frame for _ftoa is missing.
Cay you try compiling with `-O0` to disable optimizations?
Also stepping through `_ftoa` may help.
Or you can try returning from IRQ (breaking the infinite loop) - use instruction stepping, change PC to same 'BX LR' instruction and step through it (I hope this works on ARM A processor)

mpaland, 2026-10-31:
Sorry for the late answer.

The library reads floats as plain `double` with `va_arg`, as the C standard requires. The crash before `_ftoa` is executed points to the build setup: all objects, including the C library you link against, must use the same float ABI and FPU options (`-mfloat-abi=hard -mfpu=vfp`), the stack must be 8-byte aligned at every call (AAPCS), and the FPU must be enabled before the first floating point instruction, also in interrupt handlers.
The watchdog IRQ in the Cortex-M4 trace above may also point to an unhandled fault or a stack overflow, as `_ftoa` needs about 32 bytes of buffer plus a few `double` locals.

Version 4.1.0 contains several fixes in the float conversion. If it still crashes with it, please open a new issue with a minimal example and the backtrace. Closing it for now.

closed 2026-10-31 by mpaland