mpaland/printf #79: Mark the variadic *printf functions with __attribute__((format)).

svkampen, 2020-05-16:
[The `format` attribute](https://gcc.gnu.org/onlinedocs/gcc-3.2/gcc/Function-Attributes.html) marks a variadic function as taking a format string and corresponding arguments which should be type-checked against the format string. Marking printf_, sprintf_, etc. this way allows compilers to issue warnings when the supplied arguments do not have the correct type, as they would for the standard library functions.

The code change is very small, and currently implemented in https://github.com/svkampen/printf/commit/ba025a512cfb105ce89a084180312a1e1b29c1c5. I've guarded the __attribute__ declarations with `#ifdef __GNUC__ ... #endif`, so that they don't interfere with compilers that don't support them.

joshklod, 2020-07-02:
The `v*printf`-style functions can also be marked using
```
format(<archetype>, <string-index>, 0)
```
to check the format string itself for errors. See [the same link](https://gcc.gnu.org/onlinedocs/gcc-3.2/gcc/Function-Attributes.html).

svkampen, 2020-07-03:
Thanks, I must have glossed over that while reading the documentation... I've updated my branch - https://github.com/svkampen/printf/commit/d8aeaa9c625f72cc4c593a9ebeeffd244c76ae16 now contains the attributes for all of the functions.