mpaland/printf #82: Variable "va" is used before its value is set

nacansino, 2020-07-21:
I'm getting a weird build warning when building on CCRH compiler (for Renesas RH850).
Compiler is using C99 standard.

Anyone here encountering the same thing?

```bash
.\..\..\src\drivers\common\printf.c(864):W0520549:Variable "va" is used before its value is set
..\..\..\src\drivers\common\printf.c(875):W0520549:Variable "va" is used before its value is set
..\..\..\src\drivers\common\printf.c(885):W0520549:Variable "va" is used before its value is set
..\..\..\src\drivers\common\printf.c(908):W0520549:Variable "va" is used before its value is set
```

chqrlie, 2020-07-25:
There is nothing wrong with the code flagged by your compiler, and there is not much that can be done to "improve" the code.

Compiling the module as C instead of C++ might help fix this. Otherwise you might need to disable this warning specifically. 

If the compiler comes with its standard library source code, you could look at their implementation of `sprintf` to see how they avoid this warning. It is possible that the standard library be compiled with fewer warnings enabled or that they ignore such warnings in the library build process.

mpaland, 2026-10-31:
Thanks for the report, and thanks @chqrlie for the answer.

This is a false positive of the CCRH compiler: `va_start()` initializes `va` before every use, and `va_end()` is called afterwards.
Please disable that warning for `printf.c` or compile it as C. Closing, as there's nothing to change here.

closed 2026-10-31 by mpaland