mpaland/printf #89: recommendation!

tinebp, 2020-09-17:
great API!
I strongly recommend moving this into a header-only implementation to increase adoption!
check out this tool as example!
https://github.com/catchorg/Catch2

eyalroz, 2021-08-03:
This library already uses catch2.

mpaland, 2026-10-31:
Thanks for the suggestion!

The library is already just one `.c` and one `.h` file with no dependencies, so integrating it is as simple as it gets.
A header-only variant would need an extra "implementation" define and has no real advantage in C.
If you prefer it, you can `#include "printf.c"` in one translation unit. Closing, as it's not planned (despite I'm a huge fan of header-only implementations).

closed 2026-10-31 by mpaland