mpaland/printf #91: puts must append newline

aautotte, 2020-10-15:
> int puts ( const char * str );
> Write string to stdout
> Writes the C string pointed by str to the standard output (stdout) and appends a newline character ('\n').
> 
> The function begins copying from the address specified (str) until it reaches the terminating null character ('\0'). This terminating null-character is not copied to the stream.
> 
> Notice that puts not only differs from fputs in that it uses stdout as destination, but it also appends a newline character at the end automatically (which fputs does not).

https://www.cplusplus.com/reference/cstdio/puts/

vgrudenic, 2020-12-22:
I wasn't able to find the `puts` function in this repository, can you clarify your issue a bit better?

aautotte, 2021-01-04:
Sorry It was in a fork

closed 2021-01-04 by aautotte