mpaland/printf #94: Kleinere Bugs

MIchaelKopka, 2020-12-06:
Hi Marco,
gute Arbeit!
Ich habe die Funktionen auf einen PIC32 (MIPS Architektur) erprobt. 
Dabei sind mir zwei Problemchen aufgefallen.

1. In _etoa wird im Original eine 'Wandlung' uint64_t in double eingeführt.
  union {
    uint64_t U;
    double   F;
   } conv;

double ist auf einem PIC32 32 Bit lang.

Korrekt wäre hier
  union {
    uint64_t U;
    long double   F;
   } conv;

------

2. Dies ist ein schwerwiegendes Problemchen :

Am Ende von _vsnprintf wird der Ausgabestring mit 0 terminiert.

Im Original mit : 
out((char)0, buffer, idx < maxlen ? idx : maxlen - 1U, maxlen);

Hier gibt es einen Mix für maxlen als (int) und (unsigned int).
In der Original Version führt dies dazu, dass der Terminator nicht ans Ende des Strings sondern zwei Zeichen vor dem String gesetzt wird. In einem anderen Betrag wird dies, glaub ich, auch angesprochen. 
(Wenn der Ausgabe String auf dem Stack liegt gibt es nette und schwer zu findende Effekte..)
Besser wäre 
  if (idx < (unsigned int)maxlen)
    out((char)0, buffer, idx, (unsigned int)maxlen);
    else
    out((char)0, buffer, (unsigned int)maxlen - 1, (unsigned int)maxlen);

Ich habe nicht alle Beträge gelesen. Ggf. wurden die Problemchen schon angesprochen. Es kann sich hier auch um PIC32 (MIPS) spezielle Effekte handeln.

Nochmal danke für die Arbeit!

LG Michael

vgrudenic, 2020-12-22:
Regarding 2., are you referring to [this line](https://github.com/mpaland/printf/blob/d3b984684bb8a8bdc48cc7a1abecb93ce59bbe3e/printf.c#L853)? Because it seems that `idx` and `maxlen` are both `size_t` in that function.

MIchaelKopka, 2020-12-26:
Hello vgrudenic
you are absolutely right. I had a problem with my compiler and the size_t definition. 
If size_t, corresponding to C99, is correctly interpreted as "unsigned int" it works of course.
Sorry for my confusion!

eyalroz, 2021-07-14:
@MIchaelKopka : About problem #1 - if I am not mistaken, long double is not 64 bits on all platforms.

closed 2026-10-31 by mpaland