mpaland/printf #97: double with 15digits precision?

igor-m, 2021-03-17:
Hi, what is the reason to limit the precision to 9 digits only?
```
  // limit precision to 9, cause a prec >= 10 can lead to overflow errors
  while ((len < PRINTF_FTOA_BUFFER_SIZE) && (prec > 9U)) {
```
I would be happy to have double float printed with full 15digits precision - is that 9 digits limit an issue with some algorithms used? Thanks for your work!

ledvinap, 2021-03-18:
uint32_t is used for printing, so e9 is limit

igor-m, 2021-03-18:
You work with double (64bits) already - line 487:
```
  union {
    uint64_t U;
    double   F;
  } conv;

  conv.F = value;
  int exp2 = (int)((conv.U >> 52U) & 0x07FFU) - 1023;           // effectively log2
  conv.U = (conv.U & ((1ULL << 52U) - 1U)) | (1023ULL << 52U);  // drop the exponent so conv.F is now in [1,2)
```

ledvinap, 2021-03-19:
here is fractional part printing: https://github.com/mpaland/printf/blob/master/printf.c#L413

igor-m, 2021-03-19:
I added to the pow[] array the powers of up to 10^15 and changed the precision to 15, but still I get wrong results, usually the first 6 decimal places is zero, ie. like pi
3.000000237623847
etc.

igor-m, 2021-03-19:
ok, this looks better:
```
PI= 3.141592653589793e+00
A=  -1.234567898272192e+08
1 xyz = 1.000000001957684e+00
2 xyz = 1.999999999819259e+00
3 xyz = 2.999999998867262e+00
4 xyz = 4.000000000341832e+00
5 xyz = 4.999999999327491e+00
6 xyz = 5.999999999520980e+00
7 xyz = 6.999999999588305e+00
8 xyz = 7.999999999946748e+00
9 xyz = 8.999999999832568e+00
```
With uint64_t instead of long double
```
  int whole = (int)value;
  double tmp = (value - whole) * pow10[prec];
  uint64_t frac = (uint64_t)tmp;
  diff = tmp - frac;
```

eyalroz, 2021-07-14:
@ledvinap : Is there any good reason to limit this to uint32_t and 1e9? I mean, uint64_t's are used elsewhere in `printf.c` after all.

mpaland, 2021-07-14:
@eyalroz, @ledvinap Thanks so much for this massive amount of input from you guys!
Right now I'm just too busy with another project, I'll get back to printf in August. Promised.

eyalroz, 2021-07-14:
@mpaland : If you do, please email me and let's chat, as my fork currently integrates most other people's contributions...

mpaland, 2026-10-31:
The limit of 9 digits after the decimal point is a deliberate trade-off.
The fractional part is computed in an `unsigned long`, which is 32 bits on most embedded targets, and the conversion uses plain `double` arithmetic. Going to 15 or more digits would need 64-bit integer math everywhere (slow on 8/16-bit MCUs) and still wouldn't give correctly rounded digits, as your experiments show.

This is documented in the README under *Caveats*: further digits are printed as `0`. Closing as won't fix for now.

closed 2026-10-31 by mpaland