mpaland/printf #104: dos-style line breaks :-(

eyalroz, 2021-06-28:
Currently, most (but not all) files in this repository have CRLF's and end of lines - like in MS-DOS... those are quite uncommon these days. I suggest you switch to regular LF-only line breaks.

Gnarflord, 2021-07-09:
I'd suggest something along the lines of
```
char const newline[] = "\r\n";  //  Or whatever the convention is
```
I fully agree with you that CRLF-newlines are horrible and we should all just switch to `\n` but I got some test gear with very specific taste in regards to their carriage return (everything from `\r\n` to only `\n` to `\n\r` and some with just a random byte as "end of command"). So maybe have LF-only as the default but make it switchable inside of printf.h or something?

eyalroz, 2021-07-09:
@Gnarflord : Oh, no, I think you misunderstood me. I mean that the source files in this repository have DOS-style line breaks. what you're suggesting is about what `printf()`, `sprintf()` etc. _produce_.

Gnarflord, 2021-07-09:
Ah haha :D
Yeah of course, seems like I need another coffee. And here I was, trying to look at your changes and complaining about the github diff-tool being broken as it showed every single line being changed though they all look the same lol
Sorry to bother!

eyalroz, 2021-07-09:
@Gnarflord : Well, regardless - this is fixed in my [fork](https://github.com/eyalroz/printf/). I noticed @mpaland has been basically inactive for 2 years now, so I merged a bunch of PR's in there, and also switched to CMake for building.

closed 2026-10-31 by mpaland