mpaland/printf #106: Switch to a CMake-based build

eyalroz, 2021-06-28:
Instead of a Makefile which makes several assumptions it doesn't need to, consider basing the build on the CMake build system generator. This will allow:

* A convenient interface for setting compilation options (e.g. support floats or not, buffer sizes)
* Conditionally building tests
* Better cross-platform and cross-development environment compatibility
* A mechanism for installing the library as a package

ledvinap, 2021-06-30:
This project is intended to be included in other projects, it is not build as library. Using CMake for unittest is fine, but compilation options may cause problems when other 'outer' build system is used ...

eyalroz, 2021-06-30:
@ledvinap : CMake is even better for use in other projects, and I've included  a mechanism for installing and exporting. Also, you can always create a Makefile using CMake, if you really like Makefiles for some reason. And Makefile's suffer from the same problem, even worse: For example, Windows people don't use Makefile, while with CMake you can generate MSVS projects, or nmake scripts, or whatever you like. Same goes for other platforms.

Also, if someone wants to just use the two source files (`printf.h` and `printf.c`), they can do that regardless.

But - if there's a usage scenario I've missed, please elaborate...

ledvinap, 2021-06-30:
It should be fine as long as print.[ch] is not dependent on CMake use ..

eyalroz, 2021-06-30:
@ledvinap : Oh, it isn't. These two files simply exist. If you'll look at the [commit](https://github.com/eyalroz/printf/commit/69762307f64a38d37872b277db07580cfede0194), you'll see the only changes are additions of `CMakeLists.txt` files and removal of the `Makefile`.

closed 2026-10-31 by mpaland