mpaland/printf #115: Consider unifying _ntoa_long and _ntoa_long_long

eyalroz, 2021-07-31:
When `ll` is supported, why should we have both `_ntoa_long` and `_ntoa_long_long`? They do the exact same thing, except for the different types. Why not just put the value in a `long long` and use just one of the functions? It's not as though we care that much about the performance cost or stack size along the `long` path that saving 2 or 4 bytes matters so much... after all, we don't have `_ntoa_short`.

eyalroz, 2021-08-01:
Well, it's done on [my fork](https://github.com/eyalroz/printf), that wasn't so hard...

mpaland, 2026-10-31:
Thanks for the suggestion.
The split is intentional: on 8- and 16-bit targets (AVR, MSP430, STM8) a `long long` division is a large and slow library call.
Keeping `_ntoa_long` means `%d` and `%lu` never pull it in, and `PRINTF_DISABLE_SUPPORT_LONG_LONG` removes `_ntoa_long_long` completely.
The stack saving isn't the point, the code size and speed of the 32-bit path are.

closed 2026-10-31 by mpaland