mpaland/printf #116: Use a single, appropriate type for the base

eyalroz, 2021-07-31:
The only base values the library support are: 2, 8, 10, 16. And yet - we pass it around in larger types, and not even the same type everywhere.

So, why should we not just use a single small type for the base? Say, `uint_least8_t` or `uint_fast8_t`?

eyalroz, 2021-07-31:
@mickjc750 : This issue report is inspired by your commit b907595b09bc7b03ed0803f731ec5aeffa80b91d , where you've done something similar. Which of the types do you believe is more appropriate?

mickjc750, 2021-08-01:
For base, just a simple uint8_t, as there's no arithmetic performed on base
which would require masking it down to 8bits on a 32bit cpu.
For precision and width, uint_fast8_t would provide a slightly smaller (and
faster) executable than uint8_t , as there are places where precision is
decremented.

In the ntoa(), I'd recommend using literal values %2 %8 %10 %16 rather than
%base, as it saves performing a division. It's a minor size increase for a
big performance increase.

-- 
--
Michael Clift - Design Engineer.

Clift Innovations Pty Ltd
Suit 10/13 Walkers Road
Nunawading, VIC 3131

ABN 24 286188328

Tel: +61 3 9877 3109
Fax: +61 3 9999 1406
Mob: 0417 613 680
Email: ***@***.***
Web: www.clift.com.au
*****************************************************************************
PRIVATE AND CONFIDENTIAL
This e-mail and any files transmitted with it are intended solely for the
use of the addressee(s) and may contain information that is confidential or
privileged. If you receive this e-mail and you are not the addressee(s),
please disregard the contents of the e-mail, delete the e-mail and notify
the author immediately.
*****************************************************************************

eyalroz, 2021-08-01:
@mickjc750 : You've packaged several issues into one... anyway, are we sure width always fits in a uint8_t ?

mickjc750, 2021-08-01:
Not 100% sure. It really comes down to opinion as to what a reasonable
width limit is.
 Printf is generating text, 255 sounds like a reasonable limit for the
width of a number. Or the width of a field with a %s aligned in it.

mpaland, 2026-10-31:
Using `uint8_t` for the base saves IMHO next to nothing, because the value lives in a register anyway, and on some 32-bit cores it even adds masking instructions.
For width and precision, 8 bits would introduce an arbitrary limit of 255, which is not acceptable for a standard `printf` (`%300s` is valid). Closing for now.

closed 2026-10-31 by mpaland