mpaland/printf #123: Support for FreeRTOS lxip specifier

sdcsystems, 2021-08-14:
In the FreeRTOS-TCP network stack they use the custom specifier %lxip which is passed an IP address as a uint32_t and aims to print it as a xxx.xxx.xxx.xxx format.

Can this be added?

-Andy.

eyalroz, 2021-08-14:
@sdcsystems : 

The official author and maintainer of this repository, @mpaland , has been inactive for about 2 years now.

If you were to ask me this question about my active [fork](https://github.com/eyalroz/printf), I would answer negatively, because the idea is to be standard-compliant; and exceptions are very limited as well as widely-applicable, like`%t` for `ptrdiff_t`. For a platform-specific format specifier, you probably want to create your own repository based on this one. 

To change people's mind about this, you would probably need to convince them that this is a very widespread extension. Is it?

mpaland, 2026-10-31:
Thanks Andy.

`%lxip` is a FreeRTOS+TCP extension, not a standard format specifier, and I want to keep this library standard compliant.
But an intresting idea. You can print an IP address with standard specifiers:

```c
printf("%u.%u.%u.%u", (unsigned)(ip & 0xFF), (unsigned)((ip >> 8) & 0xFF),
                      (unsigned)((ip >> 16) & 0xFF), (unsigned)(ip >> 24));
```

(byte order depending on how the stack stores the address), or just wrap that in a small helper.

closed 2026-10-31 by mpaland