mpaland/printf #127: could you support %I64d %I64u like msvc done?

superzmy, 2021-09-19:
vs and gcc recognise %ld to be 32bit and 64bit. "%lld" to be 64bit.
To cross platform we aways use int32_t int64_t, so we  have to use PRI64d. It's disgusting.

eyalroz, 2021-09-28:
Hello @superzmy ,

You may have noticed Marco has not been active in this repository for over 2 years... so, you may want to consider opening this issue against [my fork](https://github.com/eyalroz/printf), which integrates [numerous bug fixes and some accuracy improvement](https://github.com/eyalroz/printf/releases/tag/v5.0.0).

mpaland, 2026-10-31:
`%I64d` is a Microsoft extension and I won't add it, as the library follows the C standard.
For portable 64-bit output please use `%lld` / `%llu` (enabled by default, see `PRINTF_DISABLE_SUPPORT_LONG_LONG`) or the `PRId64` / `PRIu64` macros from `<inttypes.h>`. `%j` (intmax_t) and `%z` (size_t) are supported, too.

closed 2026-10-31 by mpaland