mpaland/printf #128: https://github.com/eyalroz/printf  <-This repo is maintaining

mysterywolf, 2021-11-28:
https://github.com/eyalroz/printf

eyalroz, 2022-02-24:
@laoshaw : I would be glad to coordinate with mpaland, but he has been unavailable for a long period of time. By now, there are significant differences throughout the library code; and this library is not seeing any development in other directions.

Bottom line: Just use [eyalroz/printf](https://github.com/eyalroz/printf), there's nothing here you would miss.

laoshaw, 2022-02-24:
Thanks. I will switch to eyalroz/printf for now then.

bwxz, 2023-03-01:
改了不少东西啊，宏定义开关也改了

eyalroz, 2023-03-01:
Indeed, many things have been changed - to fix bugs, and address the needs of more usage scenarios.

TheClogger666, 2023-11-23:
> https://github.com/eyalroz/printf

Fork hasn't been touched it looks like. Oh well

eyalroz, 2023-11-23:
> > https://github.com/eyalroz/printf
> 
> Fork hasn't been touched it looks like. Oh well

The fork is being maintained, my friend.

TheClogger666, 2023-11-23:
> > > https://github.com/eyalroz/printf
> > 
> > 
> > Fork hasn't been touched it looks like. Oh well
> 
> The fork is being maintained, my friend.

Oh, okay! My apologies

mpaland, 2026-10-31:
This repo is now back on track!

closed 2026-10-31 by mpaland

mickjc750, 2026-11-01:
He's ALIVE!!!