mpaland/printf #131: Zero padding disabled when precision specified

Iangitpers, 2021-12-11:
An issue I've noticed: 
c=29;
printf("%03.u", c);
Output: " 29"
Expected "029"

In v4 around line 724:
 
```
       // ignore '0' flag when precision is given
        if ((flags & FLAGS_PRECISION)) {
          flags &= ~FLAGS_ZEROPAD;
        }
```
A hack:
```

       // ignore '0' flag when precision is given
        //HACK: we want the leading 0s with precision....
        if ((flags & FLAGS_PRECISION) && (*format != 'u')) {
          flags &= ~FLAGS_ZEROPAD;
        }
```

mickjc750, 2021-12-11:
"%03.u" 
 You have specified the 0 flag and a width of 3, followed by an incomplete precision (no digits after the .)
Standard behavior is to ignore the 0 flag when precision is specified.
 If you want a precision of 3, your string should be "%.3u", if you want a width of 3, your string should be "%03u", both will produce the same output for unsigned integer types.
 This repo is no longer maintained. If you want a fixed version of this repo, take a look at https://github.com/eyalroz/printf
If you want something more lightweight, you may like https://github.com/mickjc750/prnf

eyalroz, 2021-12-12:
@Iangitpers : You'll want to close this issue...

Iangitpers, 2021-12-12:
Thank you. I see that this is my error.

closed 2021-12-12 by Iangitpers