mpaland/printf #132: this is incredible

rickyelopez, 2021-12-29:
been banging my head against a HardFault due to newlib nano `snprintf` malloc call with FreeRTOS for days, this solved my issues instantly.
Beauty, thanks!

closed 2021-12-29 by rickyelopez

eyalroz, 2021-12-29:
@rickyelopez : If by "this" you mean this library, note that it has been abandoned for over 2 years; I am maintaining a  fork with many bug fixes and some missing features: https://github.com/eyalroz/printf .

rickyelopez, 2021-12-29:
oh hey thanks for pointing that out, I'll take a look!

rickyelopez, 2021-12-30:
@eyalroz your fork takes up 4 more pages of ram than this one (on my chip at least) even with all the optional functionality disabled, and more importantly, crashes my display driver.

mickjc750, 2021-12-30:
@rickyelopez 
If you want something lighter, you may like my prnf, it's not fully compatible with the standard printf though.
https://github.com/mickjc750/prnf

eyalroz, 2021-12-30:
@rickyelopez : This is interesting. I would appreciate a bug report on my repository.

rickyelopez, 2021-12-30:
> @rickyelopez If you want something lighter, you may like my prnf, it's not fully compatible with the standard printf though. https://github.com/mickjc750/prnf

I'll keep this in mind if I find myself having to shave off more space, thanks!

> @rickyelopez : This is interesting. I would appreciate a bug report on my repository.

I didn't really dig into it, but when I get a chance I can try it again and create an issue on your repo