mpaland/printf #140: I've created an Arduino Library from your source code

NormanDunbar, 2022-07-23:
Hi there @mpaland.

I don't know if you are still interested in anything using your printf code for microcontrollers, but if you are, I have created a repository at https://github.com/NormanDunbar/printf which is a fork of yours, with the added directory `ArduinoLibrary` containing the source in the required format for an installable Arduino Library and a zip file of same.

There are full instructions on my "releases" page on how to install your code into the Arduino IDE as a library.

I'm also writing a book, my second, on the Arduino and the AVR microcontroller. In the book's code, I've used your `printf` library to good effect -- thank you very much -- and thought you might like to know this also. The book is not published yet, but hopefully will be, by Apress. The working title is "Arduino Interrupts". 

Thank you very much indeed or this code and for making it available to us all. 

Regards,
Norman Dunbar.

eyalroz, 2022-07-23:
@NormanDunbar : This repository has been unmaintained for nearly 3 years now. It's been superseded by [my fork](https://github.com/eyalroz/printf).

Also, an Arduino adaptation already exists, see:
https://github.com/embeddedartistry/arduino-printf
please consider getting in touch with @phillipjohnston about this.

NormanDunbar, 2022-07-24:
Hi @eyalroz.

Thanks for the info about your fork. Strangely enough I had it open on a separate tab in Chrome! I shall convert the work I've done to use you code now. Thanks.

Thanks also for the info about @phillipJohnston's Arduino library. I was aware of that one too, but as I use PlatformIO as well as the Arduino IDE, there's a wee bug in how PlatformIO/Phillip's library structure work together which results in the `extras` directory not being picked up by PlatformIO when the library is installed. I have raised this and it's being worked on I believe.

Cheers,
Norm.

eyalroz, 2022-07-24:
@NormanDunbar : So, please close this issue (so that the link to the other repo is the first issue people see.) Feel free to file an issue against my fork, or to write me, if you need help with anything on the printf library side.

NormanDunbar, 2022-07-24:
Will do, I'm just trying yr library against an Arduino Uno sketch I'm using in my new book.

I'll close this issue here.

Cheers,
Norm.

closed 2022-07-24 by NormanDunbar